basti242/homeassistant_lg_therma_v_modbus #4: ESS Settings / Changing Energy Modes

mhbacks, 2024-10-14:
Hi Basti,

The modbus gateway is in use for more than a year and i'm able to set e.g. temperatures and so on. With the ESS Settings it is not working.

did you changed the energy modes successfully? 

I try to set it to ES1 (Operation off/Utility lock) or ES2 (Normal operation), but doesnt get a change on energy consumtion.

[image]

over the next few days, the energy supplier is installing a smart meter for a heat pump electricity rate and requires the pump to be off at a certain time.

best regards,
Max

basti242, 2024-10-14:
Hi Max, 

think ESS isnt part of my Yaml file. Can you give my some more info about your integration?
Trink german is also ok 😂?

Basti

mhbacks, 2024-10-14:
Hi Basti,
klar geht auch auf deutsch. In der Google Tabelle steht unter Register 40010 (Homeassistant 9) "Energiezustand-Eingang", dass der Wert 1 "Erzwungen Aus" ist.

Schicke ich den jetzt über eine Action zur LG, wird der Wert zwar unter `hp_energy_state_input_raw` richtig angegeben, aber die Therma V stoppt trotzdem den Betrieb nicht. Meine Frage daher ist, ob deine das auch so macht.

```
modbus:
  - name: "lg_heatpump"
    type: tcp
    host: !secret lg_heatpump_modbus_host_ip
    port: !secret lg_heatpump_modbus_port
    retries: 5
[...]
   sensors:
[...]
     - name: hp_energy_state_input_raw
        slave: !secret lg_heatpump_modbus_slave
        scan_interval: 10
        address: 9 # reg 10
        input_type: holding
```

Bei mir ist die slave id 2, nicht 1.

```
action: modbus.write_register
data:
  hub: lg_heatpump
  address: 9
  slave: 2
  value: 1
```

Die Tabelle entspricht der Doku. Das habe ich schon gegen geprüft

[image]
MFL68026607_17_231011_02_WEB-6_German.pdf

Meine momentane Anlagendokumentation, nur einmalig abrufbar. Der Link wird später entfernt.
https://pastebin.com/jPhiJXdL

basti242, 2024-10-14:
Jetzt wird es klarer. Ich schaue mir das morgen mal an.

gRiMMi83, 2024-10-16:
bei mir sieht das ähnlich aus wie bei dir, hätte auch die Hoffnung, dass ich über die Energiezustände eine konstantere Arbeitszeit ohne das auf und ab hinbekomme. Wisst ihr ob die Stromspitzen gleich mit der "Taktung" von Wärmepumpen zusehen ist? Also was ja eigentlich negativ ist wenn sie oft ein und ausschaltet?

[image]

basti242, 2024-10-16:
@gRiMMi83 
Wo hast du die Verbrauchswerte her? Extern gemessen oder habe ich übersehen, dass die Pumpe das messen kann?
Bei mir kommen die Verbrauchsdaten aus einem Shelly EM3.

Bei mir sieht es nicht ganz so "übel" aus. Vielleicht kriegen wir das aber gemeinsam optimiert und dann auch dokumentiert.
[image]

@mhbacks 
Das war aber nicht dein Thema oder?

Gruß
Basti

gRiMMi83, 2024-10-16:
ich lese die noch über das WLAN Modul aus, wenn mein Elektriker da war dann hätte ich es später über nen Shelly Pro 3EM 3-Phasen vor

mhbacks, 2024-10-16:
Hi,
ich lege mal meine Anlagendokumentation mal offen. Vielleicht hilft es auch anderen. Wir hatten diesen Januar einen extrem hohen Stromverbrauch. Dies und die schlechte Doku heben mich veranlasst, mir das Verhalten etwas genauer anzutun. Ich bin mit dem Verständnis mittlerweile weiter als das dokumentierte

https://lg.sinal.de/

@gRiMMi83 
Das Verhalten deiner Anlage lässt sich nicht so einfach erklären, aber ich versuche es mal. Zum einen hast du die Brauchwasseraufbereitung und den Heizmodus. 
Bei der Brauchwasseraufbereitung wird bei meiner Einstellung die Wärmepumpe gestartet, wenn die Temperatur des Tanksensors unter einen Schwellwert kommt. Dieser wird durch die eingestellte Solltemperatur und der Hystereseschaltung beeinflusst. 
Bei dem Heizmodus, wenn die Temperaturquelle auf Wassertemperatur steht, ist die Zirkulationspumpe in der WP immer an und springt immer dann an, wenn die Temperatur von der Solltemperatur abweicht. Ich hatte zwischenzeitlich den "AI Modus", welcher einfach nach der Heizkurve arbeitet, deaktiviert, da der Außensensor ein paar Grad abfällt, wenn die WP arbeitet und dies wieder die Solltemperatur beeinflusst. Ich habe mir damit erhofft, dass die WP so konstanter arbeitet. Kann ich aber noch keine Aussage zu treffen.
Der Stomverbrauch hat einen direkten Zusammenhang mit der momentanen Rücklauftemperatur und der Vorlauf/Rücklauf Temperaturdifferenz. Bei gleicher Temperaturdifferenz und steigender RL Temperatur steigt der Stromverbrauch und sinkt, wenn die VL Temperatur sich der RL annähert. Wenn die Differenz einen Wert unterschreitet, die die WP aus.
Jetzt nochmal die Brücke zum getakteten Betrieb: Die WP wechselt zum einen vom Heizbetrieb zur Brauchwasseraufbereitung konfigurierbar zyklisch und anschließend wird das entsprechende Programm abgearbeitet. Wie schnell die Sollwerte erreicht werden, hängt dann von dem Pufferspeicher des Brauchwassers und/oder Heizkreislauf ab.
Ich habe es aber noch nicht geschafft, die WP konstant niedrig über lange Zeit laufen zu lassen. Zum Stand heute kann ich nur sagen, dass der beste Weg zu niedrigen Heizhosten eine möglichst niedrige Vorlauftemeratur im Heizkreis benötigt. Da steckt der größte Verbrauch drin. Die Brauchwasseraufbereitung ist dabei nur ein kleinerer Teil des Verbrauchs.

Die Energiesparmodis sind auch einfach erklärt: Es wird je nach Modus die eingestellte Solltemperatur über den Modus um ein paar Grad erhöht oder gesenkt. ich konnte keine Veränderung in dem obig beschriebenen Verhalten erkennen. 

@basti242 
sorry, ich hatte die Rückfrage in deiner Nachricht im Halbschlaf nicht gelesen. ES1 bis ES4 kann man über die 230V Kontakte steuern oder ES1 bis ES8 über Modbus. Den "nicht verwenden" Modus 0 jetzt mal ausgenommen.

Grund meiner Fehlereröffnung war, dass ich die WP nicht dazu bewegen kann, in den "EVU Sperre" Modus zu wechseln. Setze ich die 1, bleibt der Verbrauch wie er ist. Nachdem ich es über Modbus nicht hinbekommen habe, aktivierte ich den physischen Kontakt (ES1 auf 1, ES2 auf 0) um die "EVU Sperre" einzulegen, leider ohne Erfolg. 

Wir haben ein Wärmepumpentarif beim lokalen Energieversorger gebucht, der eine Abschaltung der Anlage zu zwei Zeitfenstern am Tag fordert. Die Zeiten werden über einen potentialfreien Kontakt übergeben.

Am Ende suche ich also jemanden, der mir zeigen kann, ob sich seine Anlage gleich verhält wie meine oder ob ich noch weiter suchen muss...

Gruß
Max

mhbacks, 2024-10-26:
Hi,
habt ihr mal versucht, ob die "Energiezustand-Eingang" Einstellung auf 1 bei euch die WP abschaltet? Auch ob sich die Anzeige auf dem Display verändert?

```
modbus:
  - name: "lg_heatpump"
    type: tcp
    host: !secret lg_heatpump_modbus_host_ip
    port: !secret lg_heatpump_modbus_port
    retries: 5
[...]
   sensors:
[...]
     - name: hp_energy_state_input_raw
        slave: !secret lg_heatpump_modbus_slave
        scan_interval: 10
        address: 9 # reg 10
        input_type: holding
```

Bei mir ist die slave id 2, nicht 1.

```
action: modbus.write_register
data:
  hub: lg_heatpump
  address: 9
  slave: 2
  value: 1
```

Würd mich freuen, wenn ihr mir hier eine Rückmeldung geben könntet..

Gruß
Max

tailscale1, 2025-01-16:
> Hi,
> habt ihr mal versucht, ob die "Energiezustand-Eingang" Einstellung auf 1 bei euch die WP abschaltet? Auch ob sich die Anzeige auf dem Display verändert?

Schau dir mal diese Antwort an:
https://community.home-assistant.io/t/lg-therma-v-heat-pump-modbus-control/496233/68

closed 2025-02-03 by basti242

derpeter, 2026-09-02:
Moin ich hoffe es ist ok das issue wieder auf zu machen anstelle eines neuen.

Neben der Therma V haben wir noch ein ESS Home 10 PV system von LG. Die doku erwähnt zwar das die miteinander reden können viel konkreter wird es aber nicht. Da ich alle werte vom PV system im Homeassitant habe erscheint das setzen des "SG" register 40010 die sinnvollste option. Nun konnte ich hier nicht ganz raus lesen ob das bei @mhbacks nun geklappt hat und frage mich ob sich hier jemand mit einer PV basierten steuerung schon befasst hat. Konkret wäre es super gerade im Winter das Aufheizen von Heiz und Brauchwasser so weit wie möglich in die Sonnenstunden zu schieben.

mhbacks, 2026-09-02:
Hi Peter,
es hat nicht funktioniert, was aber auch nicht schlimm ist, da intern "nur" die Zieltemperatur scheinbar erhöht wird. 
Auch die manuelle Schaltung der Kontakte verhält sich eher träge.
Wenn du den Verbrauch erhöhen möchtest, empfehle ich dir die Zieltemperatur höher zu setzen und hoffen, dass die LG zeitnah drauf reagiert...
Ist kein schönes Produkt....
Gruß
Max

derpeter, 2026-09-02:
Ach schade. Bis her scheint meine ThermaV zummindest auf eine erhöhung der Brauchwasser tempertur eher zögerlich zu reagieren ich werde mal mit der Heiztemeperatur rum spielen. Kannst du mir sagen welche Kontakte du benutzt hast? Ich finde in der Doku irgend wie nicht raus welche das denn sein sollen. Hast du eine Version mit indoor unit ?

maximilianadolf, 2026-09-02:
Hi @derpeter,
Ich habe die Therma V in EVCC als Template erstellt (ist im Release) und nutze die Energiezustände über EVCC.
https://docs.evcc.io/de/heating/lg-therma-v-r290-monobloc-sg-ready/#_top

Kurz zu den Energiezuständen aus der Anleitung:
[image]

Ich würde hier den Energiezustand 6 und 7 empfehlen, da du hier deine Regelung individuell anpassen kannst. Zustand 7 wird in der Regel nicht geschalten, da es bei mir mit festem Tariff und PV kein Sperrkriterium (z.B. Tibber in der Dunkelflaute mit Strompreisen über x€) gibt.

Energiezustand 6 habe ich für mich auf 2°C auf Heizen und 10°C auf Warmwasser eingestellt.
Wenn ich jetzt von Energiezustand 2 (Normal) auf Energiezustand 6 stellst, dann schiebt sich mein Sollwert der Heizkurve +2°C und dein Sollwert Warmwasser auf +10°C.
Dein Heizkreis wird versuchen diese Heizkurve zu erreichen. Beim Warmwasser wird nur reagiert, wenn dann dein Ist-Wert auch ausreichend unter deinem neuen Sollwert liegt um die WW Hysterese zu triggern:
[image]

Bei mir sind z.B. normal 45°C und 10°C Hysterese. Habe ich jetzt schon auf 45°C geheizt und mein Energiezustand wechselt dann auf 6, dann bin ich ggf. unterhalb des Schaltwertes und bekomme erst bei einem Abfall der WW Ist Temperatur eine neue Warmwasserbereitung.

Da es zumindest bei meiner Anlage so mit der Schichtung im Speicher ist, das ich auch beim Anzeigewert 35°C noch mehr als ausreichend Warmwasser zum duschen habe, habe ich Warmwasser grundsätzlich schon mal mit Timer auf ein Zeitfenster von 13:00 bis 16:00 gelegt. Wenn also da jetzt EVCC mit PV Überschuss triggert (und nicht gerade das E-Auto diesen Überschuss verwendet), dann heizt das Wasser auf 55°C auf und ich muss ggf. einen Tag überhaupt keine Warmwasserbereitung durchführen.

Bei meinem Vater ohne E-Auto habe ich eine Automatisierung drin, die nach einer gewissen Zeit des PV Überschusses und Batteriestatus die Warmwasserbereitung auf AN stellt. Wenn PV Überschuss dann irgendwann weg ist, dann wird Warmwasserbereitung auf den Wert nach normalen Zeitplan gestellt.

Da ich in beiden Fällen nur Heizkörper und einen 100l Pufferspeicher habe, hat nur einen kleinen Speichereffekt und optimiert vor allem im Sommer und den sonnigen Übergangszeiten.

Der Energiezustand wird an der Inneneinheit angezeigt.
Die Zieltemperatur wird visuell nicht erhöht.

derpeter, 2026-09-07:
@maximilianadolf wow super das klingt ja viel vespechend. Ich hoffe noch nur mit homeassitant aus zu kommen. Da Passiert ja auch gerade richtig viel was modbus angeht. Ich werde mir dein template anschauen und gucken was ich davon in Homeassitant umsetzen kann. Eine Frage noch, wie setzt fu warmwasser auf an? Wenn ich den warmwasser betrieb aktiviere (egal ob per modbus oder display) passiert schlicht nix, jeden nacht um 3 heizt die pumpe dann den speicher allerdings auch unter den soll wert. Ich würde natürlich wie du sagen "Jetzt wasser warm machen wir haben zu viel PV"

PS: gibts ggf irgend wo schon einen besseren Ort zum austausch über die thermav und co als der github issue tracker?

mhbacks, 2026-09-07:
hi @derpeter ,
man kann nur den Modus aktivieren, aber nicht die LG direkt zum arbeiten bewegen. Wenn der Modus an ist, wird lediglich auf die Einhaltung der Schwellwerte geachtet.

[image]

Das Beispiel bei mir: ich habe die Wasseraufbereitung nur zu bestimmten Zeiten an. Erhitzt wird aber nur, wenn der Sensor am Tank einen bestimmten Wert unterschreitet. 

WWA Frischwasser: Man sieht, wenn man Warm Wasser zapft (keine Zirkualationsleitung)
W32 Power: Wenn die LG Warmwasser erzeugt
und der gelbe Balken: Wann die Warmwasseraufbereitung an ist.

Wenn man zapft und das Wasser im Tank unter 45 Grad C fällt, startet bei mir die Warmwasseraufbereitung, bis 50 Grad C erreicht ist und stoppt

Wenn du möchtest, kannst mir gern eine Mail schicken und ich gebe dir gern meine gesammelten Unterlagen dazu 

EMtXrWQ8 @sinal.de

sobald spam kommt, schalt ich die Mail wieder ab ;-)

@maximilianadolf super idee mit dem evcc! ist auf der TODO Liste!

Gruß
Max

maximilianadolf, 2026-09-07:
Es gibt einen Ort mit viel Diskussionen und Meinungen: https://www.photovoltaikforum.com/thread/228351-neue-w%C3%A4rmepumpe-lg-therma-v-r290/?pageNo=1

EVCC ist bei mir nur der Trigger für PV Überschuss und setzt gerade den Energiezustand. Wenn du alle Komponenten in HomeAssistant hast, dann kannst du dort natürlich auch einfach die Energiemodi setzen. Da hast du aus Bastis Template schon alle Sachen und die Automatisierungen vorhanden.

Zum steuern von Warmwasser nimmst du einfach den Switch hp_dhw und lässt nach Zeitplan oder PV Überschuss dein Warmwasser heizen:
```
     - name: "hp_dhw" # Warmwasserbereitung an/aus
        unique_id: "modbus.hp_dhw"
        slave: !secret lg_heatpump_modbus_slave
        address: 1
        write_type: coil
        command_on: 1
        command_off: 0
        verify:
          input_type: coil
          address: 1
          state_on: 1
          state_off: 0
```