basti242/homeassistant_lg_therma_v_modbus #10: Feature-Request - add sensors to measure produced heating

guevara777, 2024-11-07:
I´m waiting for my parts to control my own Heatpump with the code of your project. Thank you so much. Great work. 💯 

I´m looking for a way to measure the produced heat of my heat-pump so that i´m able to calculate the "Jahresarbeitszahl" (SCOP) of my heat-pump. you can do so by deviding the produced heat through the consumend energy.

I just found a way to caluculate the produced heat of my heatpump (thank you chatgpt).

To do so, you need this:

1.) a sensor with the "Vorlauftemperatur" (in-water-temperature)
2.) a sensor with the "Rücklauftemperatur" (out-water-temperature)
3.) for 1.) and 2.) a statistics sensor (mean) to flatten the curves
4.) a template-sensor with the following code:
```
{% set ruecklauf = states('sensor.rucklauf_statistiksensor') | float %}
{% set vorlauf = states('sensor.vorlauf_statistiksensor') | float %}
{% set durchfluss = states('sensor.stiebel_eltron_isg_altbau_volume_stream_wp1') | float %}
{% set delta_t = vorlauf - ruecklauf %}
{% set durchfluss_m3_h = durchfluss / 1000 * 60 %}
{% set leistung_watt = (durchfluss_m3_h * delta_t * 4.18 * 1000) %}
{{ max(0,leistung_watt) | int(default=0) }}
```

this variables have (for example) these values:
```
Vorlauf = 32.09
Rücklauf = 28.52
Delta = 3.570000000000004
Durchfluss m³/h = 1.2
Leistung in Watt 17907
```

5.) a riemann-integral-sensor to convert W into kWh
6.) utiliy-meter-sensors based on the riemann-integral-sensor to measure daily, weekly, monthly and yearly produced heat

Maybe this helps someone else, too ... :)

gRiMMi83, 2024-11-07:
Klingt interessant Versuch ich auch mal am Wochenende 👍

basti242, 2024-11-08:
Oh ja cool, wieder was fürs WE um die Langeweile zu bekämpfen ;-) Das schau ich mir auch mal an.
Danke

guevara777, 2024-11-08:
Ah, ich sehe. Du hast das gleiche "Problem" wie ich 😄

basti242, 2024-11-15:
Ich habe mir das gerade mal angesehen. Werde mal schauen ob die Werte plausibel sind und das dann einbauen. Kann ein paar Tage dauern.

gRiMMi83, 2024-11-17:
> I´m waiting for my parts to control my own Heatpump with the code of your project. Thank you so much. Great work. 💯
>
> I´m looking for a way to measure the produced heat of my heat-pump so that i´m [...]

ich hab bisschen rumgespielt bekomm es leider nicht hin. 
kannst du bitte noch die beiden Sensoren Vorlauf und Rücklauf aus deiner yaml hier rein kopieren, evtl. hab ich da schon den Fehler drin, da ich kein "statistics sensor" Typ gefunden habe? sorry für die noob fragen :-)

Danke

basti242, 2024-11-17:
Die Werte die ich bekomme sind irgendwie komisch. Lass mich mal noch ein bisschen testen.

gRiMMi83, 2024-11-19:
wo kommen die beiden Werte her bzw. was sagen die aus?

{% set durchfluss_m3_h = durchfluss / 1000 * **60** %}
{% set leistung_watt = (durchfluss_m3_h * delta_t * **4.18** * 1000) %}

Durchfluss / 1000 is klar aber warum mal 60?
selbes bei der 4,18 ?

basti242, 2024-11-19:
4,186 kJ/kg ist die spezifische Wärmekapazität von Wasser.
die 60 wäre in dem Fall die Umrechnung von h in min. Also in Verbindung mit der /1000 wären das L/min.

earlybite, 2024-12-23:
Ich habe das jetzt mal implementiert: https://pastebin.com/YRnJkC4V

Es wird nun die gesamte erzeugte Wärmemenge berechnet, außerdem die gesamte aufgewendete elektrische Energie. Daraus kann man dann SCOPs und die JAZ ermitteln. Zusätzlich habe ich das in meinem Fall noch für die Heizkörper und den WW-Speicher getrennt, sodass ich sozusagen einen SCOP für die Heizkörper und einen für den WW-Tank habe. 

So sieht das nun beispielhaft bei mir aus:
[img]https://up.picr.de/49062539bq.png[/img]

[img]https://up.picr.de/49062540ar.png[/img]

earlybite, 2024-12-23:
Ich habe mir jetzt die Zahlen nochmal genau angeschaut und auch mit der Energieanzeige am LG Terminal verglichen. Irgendwas scheint da noch nicht so ganz zu stimmen...

Hier ist die Ausgabe des LG-Terminals zu sehen:
[image]
[image]

Im Vergleich dazu hier meine HA-Energiekarte:
[image]

Auffällig sind mehrere Dinge: 

1. Heizen und Brauchwasser ergeben insgesamt 10,5 kWh elektrisch, das scheint sich zu decken mit der HA-Berechnung. Allerdings stimmen die Werte für Brauchwasser und Heizkörper leider nicht damit überein. Ich habe in HA einen Sensor definiert, der 1 ausgibt, sobald der Brauchwasser-Erwärmungsmodus aktiviert ist, sonst 0. Einen entsprechenden Sensor habe ich auch für den Heizmodus (Heizkörper) implementiert. Mit diesen Sensorwerten multipliziere ich den Wert des el. Leistungssensors der WP, um die el. Leistung nach Brauchwasser und Heizkreis zu trennen. Entsprechend integriert ergeben sich die hier dargestellte Werte.
2. Die gesamte Wärmemenge passt auch nicht so richtig, denn 36 kWh + 16,56 kWh ergeben ja nicht 39,54 kWh, ganz zu schweigen davon, dass LG im Terminal 34 kWh ausgibt. Ich hatte schon überlegt, dass die Wärmeabgabe der Heizkörper vielleicht zu groß berechnet wird, weil sich die HK nach dem Erwärmen des Brauchwassers die restliche Wärme der VL-Temperatur "schnappen", ohne dass dafür elektrisch geheizt werden muss. Sollte also besser die Wärmeabgabe der HK als "Wärmeabgabe Gesamt - Wärmeabgabe DHW" berechnet werden?
3. 39,54 (Wärmeabgabe gesamt) - 5,15 (elektrische Energie DHW heute) = 34,39 kWh , also ca. 35 kWh... Zufall?? Oder rechnet vielleicht die LG-Wärmepumpe einfach "Mist" aus und gibt den dann im LG-Terminal falsch aus?

Macht es vielleicht einen Unterschied, wie integriert wird? Ich habe für alle Integrale die Methode "trapezoidal" verwendet. Wäre "left" hier eine bessere Wahl?

########### INTEGRALE ###############

### Integration für Gesamtenergie ###

  - platform: integration
    source: sensor.hp_heizleistung
    name: "HP Energieverbrauch Heizung"
    unit_prefix: k
    round: 2
    method: trapezoidal
    
  - platform: integration
    source: sensor.hp_energy_instantaneous_total_high
    name: "HP Elektrische Energie"
    unit_prefix: k
    round: 2
    method: trapezoidal

### Integration für Heizkörper-Energie ###

  - platform: integration
    source: sensor.hp_heizleistung_heizkoerper
    name: "HP Energieverbrauch Heizkoerper"
    unit_prefix: k
    round: 2
    method: trapezoidal
    
  - platform: integration
    source: sensor.hp_elektrische_leistung_heizkoerper
    name: "HP Elektrische Energie Heizkoerper"
    unit_prefix: k
    round: 2
    method: trapezoidal

### Integration für DHW-Energie ###

  - platform: integration
    source: sensor.hp_heizleistung_dhw
    name: "HP Energieverbrauch DHW"
    unit_prefix: k
    round: 2
    method: trapezoidal
    
  - platform: integration
    source: sensor.hp_elektrische_leistung_dhw
    name: "HP Elektrische Energie DHW"
    unit_prefix: k
    round: 2
    method: trapezoidal

earlybite, 2024-12-25:
Ich war in Feiertagslaune und habe noch ein bisschen herumprobiert. Es lag wohl tatsächlich an der Integrationsmethode. Statt trapezoidal muss left gewählt werden, dann kommen stimmige Werte heraus. 

[image]
[image]

`type: entities
entities:
  - type: custom:mushroom-template-card
    primary: Elektrische Energie und Wärmeabgabe
  - type: custom:bar-card
    positions:
      icon: "off"
      indicator: inside
      name: inside
    width: 80%
    height: 25px
    text-align: right
    padding-left: 4px
    bar-card-border-radius: 2px
    entities:
      - entity: sensor.hp_daily_energy
        name: Wärmeabgabe Gesamt heute
      - entity: sensor.hp_daily_electric_energy
        name: Elektrische Energie Heute
    entity_row: false
  - type: section
  - entity: sensor.hp_scop_daily
    type: custom:multiple-entity-row
    show_state: false
    name: SCOP Gesamt
    icon: mdi:flash
    secondary_info: heute
    entities:
      - entity: sensor.hp_scop_daily
        name: SCOP Gesamt heute
  - type: section
  - type: custom:bar-card
    positions:
      icon: "off"
      indicator: inside
      name: inside
    width: 80%
    height: 25px
    text-align: right
    padding-left: 4px
    bar-card-border-radius: 2px
    entities:
      - entity: sensor.hp_daily_energy_heizkoerper
        name: Wärmeabgabe Heizkörper heute
      - entity: sensor.hp_daily_electric_energy_heizkoerper
        name: Elektrische Energie HK Heute
    entity_row: false
  - entity: sensor.hp_scop_daily_heizkoerper
    type: custom:multiple-entity-row
    show_state: false
    name: SCOP Heizkörper
    icon: mdi:flash
    secondary_info: heute
    entities:
      - entity: sensor.hp_scop_daily_heizkoerper
        name: SCOP Heizkörper heute
  - type: section
  - type: custom:bar-card
    positions:
      icon: "off"
      indicator: inside
      name: inside
    width: 80%
    height: 25px
    text-align: right
    padding-left: 4px
    bar-card-border-radius: 2px
    entities:
      - entity: sensor.hp_daily_energy_dhw
        name: Wärmeabgabe DHW heute
      - entity: sensor.hp_daily_electric_energy_dhw
        name: Elektrische Energie DHW Heute
    entity_row: false
  - entity: sensor.hp_scop_daily_dhw
    type: custom:multiple-entity-row
    show_state: false
    name: SCOP DHW
    icon: mdi:flash
    secondary_info: heute
    entities:
      - entity: sensor.hp_scop_daily_dhw
        name: SCOP DHW heute
`

und 
`type: custom:apexcharts-card
header:
  show: true
  title: Energieverbrauch (täglich) in kWh
graph_span: 7d
span:
  start: day
  offset: "-6d"
now:
  show: true
  label: Jetzt
series:
  - entity: sensor.hp_daily_electric_energy_left
    name: Elektrisch
    type: column
    color: blue
    group_by:
      duration: 1d
      func: diff
  - entity: sensor.hp_daily_energy_left
    name: Wärme Gesamt
    type: column
    color: red
    group_by:
      duration: 1d
      func: diff
  - entity: sensor.hp_daily_energy_heizkoerper_left
    name: Wärme Heizkörper
    type: column
    color: orange
    group_by:
      duration: 1d
      func: diff
  - entity: sensor.hp_daily_energy_dhw_left
    name: Wärme DHW
    type: column
    color: turquoise
    group_by:
      duration: 1d
      func: diff
`

Folgendes muss an die configuration.yaml angehängt werden:
https://pastebin.com/FQag2eDh

Die folgende Datei modbus_lg_heatpump_energy.yaml benutze ich zusätzlich zu bastis yaml, um die Energie-Sensoren einzulesen:
https://pastebin.com/snSG8CyF

Dieses Register muss der modbus_lg_heatpump.yaml von basti noch hinzugefügt werden, damit es funktioniert. Funktioniert scheinbar nur mit aktueller Firmware, bitte ausprobieren.

`      - name: "hp_energy_instantaneous_total_high" # momentaner Energieverbrauch high
        unique_id: "modbus.hp_energy_instantaneous_total_high"
        precision: 1
        scan_interval: 10
        address: 37
        slave: !secret lg_heatpump_modbus_slave
        unit_of_measurement: W
        input_type: input`

Viel Spaß!

cjungde, 2024-12-26:
@earlybite Danke für deine Konfig. Ich baue es gerade bei mir ein.
Ist meine Annahme richtig, das der Wert 'hp_energy_instantaneous_total_high' ein Stromzähler für die Wärmepumpe ist ? Bei mir ist das ein externer SDM72D. Oder hast Du den Wert per Modbus aus der Wärmepumpe gezogen ?
Welche anderen Werte kommen von anderen Geräten ? Ich hab erstmal keine anderen gefunden.

earlybite, 2024-12-26:
> @earlybite Danke für deine Konfig. Ich baue es gerade bei mir ein. Ist meine Annahme richtig, das der Wert 'hp_energy_instantaneous_total_high' ein Stromzähler für die Wärmepumpe ist ? Bei mir ist [...]

Ich habe den Wert per Modbus aus der WP gezogen. Es handelt sich dabei um ein neues Modbusregister, das in der Config von @basti242 noch nicht drin ist (siehe oben, Adresse 37, das muss noch in die Datei von basti eingefügt werden). Falls du die entsprechende Firmware noch nicht hast, musst du den Strom extern irgendwie messen, zB mit deinem SDM72D. Ich werde dazu demnächst einen Shelly Pro 3EM einbauen, der misst eventuell genauer. 

Ein "Problem" scheint es noch zu geben. Ich habe mal meine berechneten Werte (u.a. mit den Formeln von @ricsch berechnet) mit denen verglichen, die auf dem LG-Terminal zu sehen sind:
[image]
[image]
[image]

Die Stromwerte passen, aber der berechnete Wert für die Wärmeabgabe weicht vom LG-Wert stark ab: 37,9 kWh vs. 48,43 kWh. Weiß jemand woran das liegen könnte? Wieso kommen die von LG auf den viel kleineren Wärmewert? Die Formel von @ricsch scheint mir richtig zu sein, ich benutze sonst auch nur Werte, die die WP selbst ausgibt... Rechnen die von LG falsch???

tailscale1, 2025-01-16:
> Ich habe den Wert per Modbus aus der WP gezogen. Es handelt sich dabei um ein neues Modbusregister, das in der Config von [@basti242](https://github.com/basti242) noch nicht drin ist (siehe oben, [...]

Ich würde meinen, die Wärmeabgabe sollte nur berechnet werden, wenn die Wasserpumpe läuft. 
Das könnte den Unterschied zu den LG-Werten erklären.

maximilianadolf, 2025-08-07:
> Ich würde meinen, die Wärmeabgabe sollte nur berechnet werden, wenn die Wasserpumpe läuft. Das könnte den Unterschied zu den LG-Werten erklären.

Was sind so die bisherigen Langzeiterfahrungen mit der Berechnung?
Ich habe es nur gestern mal implementiert und mein Warmwasser "konsumiert" auch Wärmeenergie, wenn sie nicht läuft :-).
Nachdem der 0-Durchfluss auf dem Durchflusssensor leider bei 5 l/min liegt (https://github.com/basti242/homeassistant_lg_therma_v_modbus/issues/17#issuecomment-2622826144) wird halt auch im Stillstand weiter integriert.

Ich habe jetzt mal die Pumpe mit aufgenommen:
`{% set pumpe = 1 if is_state('binary_sensor.hp_water_pump_status', 'on') else 0 %}`