basti242/homeassistant_lg_therma_v_modbus #14: Weitere Modbus Register

gRiMMi83, 2024-11-20:
hab hier was neues bekommen, 

hab aber noch ein Verständnis Problem wie du die Register eingepflegt hast?

Beispiel:

Outdoor Air temp. steht im Register Input 0x04 30013

deine Abfrage

`      - name: "hp_outside_temp" # Außentemperatur
        precision: 1
        scan_interval: 10
        address: 12
        slave: !secret lg_heatpump_modbus_slave
        scale: 0.1
        device_class: temperature
        unit_of_measurement: "°C"
        input_type: input`

Warum Adresse 12? und wo wird auf das Input register verwiesen? 

hier die neue Modbus Übersicht von LG

[image]
[image]

basti242, 2024-11-20:
Homeassistant beginnt wohl mit 0 bei den Registernummerierungen und LG mit 1. Keine Ahnung wer mehr recht hat.
Der Register Type wird über den input_type oder write_type angesprochen.
Zum Teil ist es schon sehr unübersichtlich. Deshalb hab ich das hier auch mal alles dokumentiert da ich mich auch sehr schwer getan habe.
Die neuen Register werde ich mal am WE einpflegen. Leider funktionieren die nur bei neueren Heizungen.

gRiMMi83, 2024-11-20:
angeblich funktionieren die auch bei älteren Geräten seit der Firmware von September 2024

basti242, 2024-11-20:
OK und wie komme ich an die Firmware?

earlybite, 2024-12-21:
Hallo,
ich war mal so frei und habe die neuen Energie-Register (außer Kühlen) und das Register für Wasserdruck hinzugefügt. 

`# Home Assistant LG Heatpump integration
# last update: 2024-10-02 (for Homeassistant update 2024.10)

modbus:
  - name: "lg_heatpump"
    type: tcp
    host: !secret lg_heatpump_modbus_host_ip
    port: !secret lg_heatpump_modbus_port

    #------------------------------------------
    sensors:
      # Technische Daten
      # Input Register
      #------------------------------------------

      - name: "hp_error_code"
        unique_id: "modbus.hp_error_code"
        scan_interval: 10
        address: 0 # reg 1
        slave: !secret lg_heatpump_modbus_slave
        input_type: input

      - name: "hp_odu_operation_cycle"
        unique_id: "modbus.hp_odu_operation_cycle"
        scan_interval: 10
        address: 1
        slave: !secret lg_heatpump_modbus_slave
        input_type: input

      - name: "hp_inlet_temp"
        unique_id: "modbus.hp_inlet_temp"
        scale: 0.1
        precision: 1
        scan_interval: 30
        address: 2 # reg 3
        slave: !secret lg_heatpump_modbus_slave
        unit_of_measurement: °C
        device_class: temperature
        input_type: input

      - name: "hp_outlet_temp"
        unique_id: "modbus.hp_outlet_temp"
        scale: 0.1
        precision: 1
        scan_interval: 30
        address: 3 # reg 4
        slave: !secret lg_heatpump_modbus_slave
        unit_of_measurement: °C
        device_class: temperature
        input_type: input

      - name: "hp_backup_heater_outlet_temp"
        unique_id: "modbus.hp_backup_heater_outlet_temp"
        scale: 0.1
        precision: 1
        scan_interval: 30
        address: 4 # reg 5
        slave: !secret lg_heatpump_modbus_slave
        unit_of_measurement: °C
        device_class: temperature
        input_type: input

      - name: "hp_dhw_water_temp"
        unique_id: "modbus.hp_dhw_water_temp"
        scale: 0.1
        precision: 1
        scan_interval: 30
[... 890 lines cut ...]
        data_template:
          hub: lg_heatpump
          slave: !secret lg_heatpump_modbus_slave
          address: 8
          value: "{{ states('input_number.hp_dhw_target_temperatur') |float *10 | int}}"
    mode: single

  - id: "hp_automation_set_target_temperatur_input_slider_update"
    alias: "hp_automation_set_target_temperatur_input_slider_update"
    description: "Änderung der DHW Solltemperatur auf Input Slider schreiben"
    triggers:
      - trigger: state
        entity_id:
          - sensor.hp_dhw_target_temp
    condition: []
    action:
      - action: input_number.set_value
        target:
          entity_id: input_number.hp_dhw_target_temperatur
        data:
          value: "{{ states('sensor.dhw_target_temp') }}"
    mode: single
`

Beste Grüße

ricsch, 2024-12-23:
Hi @earlybite, 

wie sieht deine Modbus-Config für 'sensor.hp_energy_instantaneous_total_high' aus? Ich bekomme das leider nicht konfiguriert :-(

VG
Rico

earlybite, 2024-12-23:
Hier die eingefügten neuen Register: https://pastebin.com/XEAsEuUe

ich hab sie hinter dem sensor hp_compressor_rpm eingepflegt.

VG

ricsch, 2024-12-23:
Vielen Dank für die schnelle Antwort! Interessanterweise geht das bei mir nicht, obwohl ich die Leistungsdaten im Display ablesen kann. Ich habe eine HM071MR U44. Ich habe das doofe Gefühl, das ich um den WLAN-Stick wg. dem Firmware-Update nicht herum kommen werde.

earlybite, 2024-12-23:
hmm Mist :-/ Bei mir scheint die neue Firmware wohl schon drauf zu sein, hab die LG THerma V R32 5 kW seit November.

gRiMMi83, 2024-12-26:
> Hier die eingefügten neuen Register: https://pastebin.com/XEAsEuUe
> 
> ich hab sie hinter dem sensor hp_compressor_rpm eingepflegt.
> 
> VG

bekomme leider keine Werte

[image]

hab ne LG Therma V R32 16kW (Modell: ZHBW168A1, Type: hm163mr.u34, Firmware: SAA42121706.0000CB1A.0)

Gyr0sK0mplett, 2025-01-01:
Meine ist auch schon über 2 Jahre alt, FW 3066. Denke mal das wird nichts, alle neuen nicht verfügbar.

isarrider, 2025-01-05:
> Hier die eingefügten neuen Register: https://pastebin.com/XEAsEuUe
> 
> ich hab sie hinter dem sensor hp_compressor_rpm eingepflegt.
> 
> VG

Mach doch gleich n Pull für Basti auf ;)

isarrider, 2025-01-05:
Wenn man noch so sucht, was da sinnvolles rauskommen könnte,
stößt man auf:
https://community.home-assistant.io/t/lg-therma-v-heat-pump-modbus-control/496233/79
bzw https://gathering.tweakers.net/forum/list_message/76531486#76531486
Da sind zB diese drin:
30014
30015
30016
30017 Pipe in temp [0.1 °C ×10]
30018 Pipe out temp [0.1 °C ×10]
30019 Suction temp [0.1 °C ×10]
30020 INV1 dis. temp [0.1 °C ×10]
30021 HEX temp [0.1 °C ×10]
30022
30023 HighPress [mBar]
30024 LowPress [mBar]
30025 Inverter [Hz]

Kommt da bei der R290 was sinnvolles raus?

basti242, 2025-01-06:
Ich habe es mal am Beispiel vom Register 30025 getestet. Bei meiner R290 kommt da nix. 
Ich habe die Firmware 3.0.6.7a.

isarrider, 2025-01-13:
Ok, danke.
Firmware Update gibts auch nicht?
Man muss mit dem leben, was drauf ist?

basti242, 2025-01-13:
Ich muss mal schauen was das komische Cloud Gateway kann. Das habe ich noch nie genauer angesehen. Vielleicht gibts da auch Firmware Updates.

isarrider, 2025-01-13:
Ah, das plane ich gar nicht zu kaufen...
Regelst du eig je nach PV Einstrahlung die WP bzw geht das?

isarrider, 2025-01-29:
Also ich meine über eine angepasste Vorlauftemperatur oÄ...

basti242, 2025-01-29:
Aktuell regel ich das über den Strompreis. Wenn teuer dann Vorlauftemperatur runter.

so z.B. 

```yaml
alias: Vorlauftemperatur Wärmepumpe nach Strompreis
description: ""
triggers:
  - trigger: state
    entity_id:
      - binary_sensor.most_expensive_hours_heatpump
    id: Teuerste Stunden Wärmepumpe "True"
    to: "on"
  - trigger: state
    entity_id:
      - binary_sensor.most_expensive_hours_heatpump
    id: Teuerste Stunden Wärmepumpe "False"
    to: "off"
conditions: []
actions:
  - choose:
      - conditions:
          - condition: trigger
            id:
              - Teuerste Stunden Wärmepumpe "True"
        sequence:
          - action: input_number.min
            metadata: {}
            data: {}
            target:
              entity_id: input_number.hp_shift_value_in_auto_mode_circuit2
        alias: Vorlauftemperatur reduzieren wenn Strom teuer
      - conditions:
          - condition: trigger
            id:
              - Teuerste Stunden Wärmepumpe "False"
        sequence:
          - action: input_number.set_value
            metadata: {}
            data:
              value: 0
            target:
              entity_id:
                - input_number.hp_shift_value_in_auto_mode_circuit2
        alias: Vorlauftemperatur zurücksetzen wenn Strom normal
mode: single
```

earlybite, 2025-01-31:
> hab hier was neues bekommen,
>
> hab aber noch ein Verständnis Problem wie du die Register eingepflegt hast?
>
> Beispiel:
>
> Outdoor Air temp. steht im Register Input 0x04 30013
>
> deine Abfrage
>
> ` - name: [...]

Hallo @gRiMMi83 
Darf ich fragen, wo du dieses Bild mit den neuen Registern gefunden hast? Ich habe es geschafft, das letzte Register 40024 "Compressor Modulation" zu implementieren. Es suggeriert ja, dass man damit die Wärmepumpe auf eine bestimmte Prozentzahl "drosseln" bzw. modulieren kann. Ich habe verschiedene Werte versucht, 60 oder 50 oder 40, jedoch ist nichts nennenswertes passiert. Weißt du, wo man genauere Informationen zum Sinn und Zweck dieses Registers bekommen könnte?

isarrider, 2025-02-03:
Hast du 30018 bewußt rausgelassen?

isarrider, 2025-02-03:
Ich kann leider noch nix testen, da meine WP erst dieses Jahr kommt...
Aber ich wollte zumind mal die SW vorbereiten...

basti242, 2025-02-03:
Dann schreib dir mal einen Merker und teste bitte wenn sie da ist.

earlybite, 2025-02-04:
> Hast du 30018 bewußt rausgelassen?

Also ich habe mal das hier hinzugefügt:
      - name: "hp_pipe_out_temp" # Pipe out temp [0.1 °C ×10]
        unique_id: "modbus.hp_pipe_out_temp"
        scan_interval: 10
        precision: 1
        scale: 0.1
        address: 17
        unit_of_measurement: °C
        slave: !secret lg_heatpump_modbus_slave
        input_type: input

Da kommt bei mir jetzt seit 11 Minuten nur 1200 °C als Wert raus. Scheint also kein sinnvolles Register bei meiner R32 Anlage zu sein. 

Generell sind die Bezeichnungen bei dir und bei basti andere:
30017 Pipe in temp [0.1 °C ×10]  -> Basti: Adresse 16, hp_temp_liquid_gas
30018 Pipe out temp [0.1 °C ×10] -> Basti: Adresse 17, gibt es nicht
30019 Suction temp [0.1 °C ×10] -> Basti: Adresse 18, hp_temp_suction
30020 INV1 dis. temp [0.1 °C ×10] -> Basti: Adresse 19, hp_temp_heatgas
30021 HEX temp [0.1 °C ×10] -> Basti: Adresse 20, hp_temp_before_vaporiser
30022 (gibt es nicht?) -> basti: Adresse 21, hp_temp_after_vaporiser
30023 HighPress [mBar] -> basti: Adresse 22, hp_high_pressure
30024 LowPress [mBar] -> basti: Adresse 23, hp_low_pressure
30025 Inverter [Hz] -> basti: Adresse 24, hp_compressor_rpm

isarrider, 2025-02-16:
Danke - ja, LG hat da verschiedene Dokus...
Allerdings hätte ich erwartet, daß ein Register (30018), das nicht benutzt wird,
auch nichts liefert bzw einen Fehler wirft...

isarrider, 2025-04-02:
@earlybite kommt eig auch was raus, wenn du ein sicher nicht verwendetest Register pollst?

earlybite, 2025-04-02:
30018 bleibt bei mir so wie es basti eingestellt hat konstant auf 1200°C

TheNablaOperator, 2025-04-03:
> Hallo [@gRiMMi83](https://github.com/gRiMMi83) Darf ich fragen, wo du dieses Bild mit den neuen Registern gefunden hast? Ich habe es geschafft, das letzte Register 40024 "Compressor Modulation" zu [...]

@gRiMMi83 @earlybite für welche Type gilt diese Beschreibung der Register? Für meine Type (R32 Mono 7kW) sind diese Register nicht enthalten und Lt. LG daher nicht offiziell...?! Hängt das vielleicht mit der Type oder eher der Firmware zusammen? Habt ihr vielleicht Ideen? MFG

isarrider, 2025-11-10:
changes are I added the missing registers and swapped the ENERGY_* high and low and some cosmetics...
Careful, as this is made for Modbus RTU

dxfinfo, 2025-11-19:
> Register 30051 und 30052 stehen da aber nicht drin ..... 
> 
>

Ich habe die Adressen bis 30075 physisch gescannt. Die Adressen 47-50 sind mir noch immer unklar. Vergleich der Änderungen in der Gruppe von 35 bis 52 – was mit den Energieindikatoren zusammenhängt. 51-52 stimmen mit den Informationen der LG Heat Output-Leiste überein, 35,36 Werte mit 37,38 (möglicherweise unterscheiden sie sich, wenn bestimmte Geräte aktiviert werden, ich habe eine reine Verbindung mit Temperaturkontrolle am Ausgang).

dxfinfo, 2025-11-19:
> könnte vielleicht wie beim Strom das ganze nochmal aufgeteilt sein nach Wärme gesamt, Wärme Warmwassererzeugung usw.

Das ist gut möglich – ich nutze die Funktionen „Cool“ und „DHW“ nicht, daher kann ich das nicht überprüfen.
Auf dem Foto ist der Wert bereits im Real-Format angegeben.
Ich habe die Firmware 3.0.69.
[image]

dxfinfo, 2025-11-19:
> > Ich hab grad mal 50 (Reg 51 u 52 hinzugefügt, das erscheint ganz passend)
> 
> Haben Sie sich mit den Registern 47, 48, 49, 50 vertraut gemacht?

Präzisierung: Ich schreibe 47, 48, 49, 50, meine aber 30047, 30048, 30049, 30050 ohne Verschiebung um 0 für HA.
Denn bei Siemens gebe ich die Register für Aktionen in diesem Format an.

isarrider, 2025-11-19:
46 als COP sieht eig gut aus - hast du n längeren Graphen?
48 scheint mir auch gut getippt...

isarrider, 2025-11-19:
Du schickst ein Paket (die Zeit) und der WR antwortet mit allem, was er hat...

isarrider, 2025-11-19:
Habs mal alles geupdated und teste morgen...

isarrider, 2025-11-20:
[image]

isarrider, 2025-11-20:
COP passt hier total nicht...
modbus.hp_ENERGY_INSTANTANEOUS_TOTAL_SYSTEM und modbus.hp_ENERGY_INSTANTANEOUS_TOTAL sind fast immer gleich
Mitte gelb (hp_ENERGY_INSTANTANEOUS_TOTAL_TH - TH für Thermisch) passt auch nicht...
Toughts?

dxfinfo, 2025-11-20:
> COP passt hier total nicht... modbus.hp_ENERGY_INSTANTANEOUS_TOTAL_SYSTEM und modbus.hp_ENERGY_INSTANTANEOUS_TOTAL sind fast immer gleich Mitte gelb (hp_ENERGY_INSTANTANEOUS_TOTAL_TH - TH für Thermisch) passt auch nicht... Toughts?

Ja, - 46 (30047) ist unklar, aber 48 (30049) ist möglicherweise die momentane Wärmeabgabe analog zu  36 (30035) ... (es lässt sich das gleiche Verhältnis wie bei 38 (30037) und 50 (30051) feststellen, könnte das sein?

isarrider, 2025-11-20:
48 - nein, das passt nicht, das ist im mittleren Graph die gelbe Linie...

userwithoutpassword, 2025-11-20:
> aber 48 (30049) ist möglicherweise die momentane Wärmeabgabe

Das könnte bei mir passen. Habe aber bei der Warmwassererzeugung dort Werte stehen die etwas über der Nennleistung meiner Wärmepumpe sind.

46 kann nicht der COP sein, wie erst angenommen. Bei der Trinkwassererwärmung steigt dieser Wert auf über 100 und bleibt auch danach noch einige Zeit so hoch. Bei der Enteisung steigt er sogar über 6000, teilweise sogar 7000

isarrider, 2025-11-20:
Von heute:

[image]

isarrider, 2026-10-03:
Ok, a longer discussion with Claude and Gemini: It seems to be a Factor that is prob going into the Heat Demand and as such in the  Modulation of the Compressor... Naive theory is it goes from 0 to 120% and tells the Compressor how fast it should spin - 120% could mean the heatpump would be meant to run at 100% max constantly and 120% is for Peaks - like my 12kW is meant to run max at 10kW over a longer time...

isarrider, 2026-10-03:
I will call it DvS (Demand vs Supply), until sb comes up with a better explanation...

isarrider, 2026-10-03:
- name: "hp_Demand_vs_Supply"
        unique_id: "modbus.hp_DvS"
        address: 46 # Input Register 47: Heatpump internal calculated Demand vs current Supply
        input_type: input
        slave: 1
        data_type: uint32
        scale: 1
        precision: 2
        unit_of_measurement: "%"
        scan_interval: 10