basti242/homeassistant_lg_therma_v_modbus #16: Feature-Request - Force DHW heating mode

earlybite, 2024-12-30:
Hi Basti,

Thanks again for the great guide and all the valuable input. I’ve put together a nice dashboard for my LG Therma V R32 5 kW Monobloc S. However, there’s one thing I’m still missing: a switch to force the heating of the domestic hot water (DHW) tank.

Example Scenario
The heat pump operates in heating mode for radiators and heats the DHW only via Home Assistant control, e.g., once in the morning at 5 a.m.

Currently, to my knowledge, it’s only possible to disable the DHW heating mode for certain periods. However, turning on the DHW heating mode doesn’t mean the hot water will be heated immediately. The timing of the heating process seems to depend on the settings in the installer mode under “DHW Time Setting.”

If you turn the DHW heating mode off and back on, it doesn’t automatically initiate the DHW heating process.

My Question
Is there a workaround or a Modbus switch that could be integrated to force the DHW heating process?

From what I’ve observed, the only thing that can trigger the DHW heating process is the disinfection mode, which starts both the DHW heating element and the DHW heating process simultaneously (though it’s not very economical).

Thanks in advance, and best regards!

===

Hallo Basti, Danke nochmals für die tolle Anleitung und die vielen Inputs. Ich habe mir mittlerweile ein schönes Dashboard zu meiner LG Therma V R32 5 kW Monobloc S zusammengebaut. Nur eins fehlt mir dabei noch: Ein Schalter, mit dem man das Aufheizen des Warmwassertanks erzwingen kann. 

Beispiel-Szenario: Die WP ist nur im Heizkörper-Betrieb und erwärmt das Brauchwasser nur durch Home Assistant gesteuert, z.B. einmal morgens um 5 Uhr. 

Aktuell ist es meines Wissens nach nur möglich, den Brauchwasser-Heizen Modus für gewisse Zeitspannen auszuschalten. Wenn der Brauchwasser-Heizen Modus aber eingeschaltet wird, heißt das nicht, dass das Brauchwasser sofort erwärmt wird. Wann die Erwärmung stattfindet, hängt, soweit ich das überblicken kann, von den Einstellungen im Monteur-Modus "Trinkwasser Zeiteinstellung" ab. Wenn man den DHW-Heizen-Modus aus- und wieder einschaltet, wird also nicht automatisch das Brauchwasser-Heizen initiiert. 

Gibt es für dieses Erzwingen des DHW-Heizvorgangs einen Workaround oder einen Modbus-Schalter, den man noch einbinden könnte? Das einzige, was den DHW-Heizmodus triggern kann, ist meines Wissens der Desinfektionsbetrieb, bei dem der Brauchwasser-Heizstab und der DHW-Heizvorgang gleichzeitig gestartet werden (auch eher unwirtschaftlich...). 

Danke und viele Grüße!

basti242, 2025-01-04:
Sorry, hab dich vergessen. 

Wenn du DHW erhitzen einschaltest, sollte es auch sofort losgehen. Was die Heizung ggf. daran hindert könnte die eingestellte Hysterese sein?

Wie sind die Einstellungen bei Dir?

earlybite, 2025-01-05:
Die Hysterese ist auf den Wert 3 eingestellt.

basti242, 2025-02-03:
Sorry ist irgendwie untergegangen.
Wie sind denn die DHW spezifischen Einstellungen in deiner Heizung? Dann könnten wir mal vergleichen.

earlybite, 2025-02-04:
Also ich habe eingestellt: 
Brauchwasser Sollwerttemperatur min 40, max 55
Desinfektion deaktiviert (bisher noch kein Heizstab verbaut)
Zustand des Warmwasserspeichers: min 5, max 55
Info Warmwasserspeicher: Hysterese 3, Priorität Brauchwasser
Priorität der Heizung: Heizung + Zusatzheizung
Brauchwasser Zeiteinstellung: 60 , 30
Umwälzzeit: N/A

earlybite, 2025-02-04:
Die Soll-Temperatur für DHW ist bei mir übrigens auf 55 °C eingestellt.

Hier auch noch eine Grafik zu einem beispielhaften DHW-Verlauf bei mir:

[image]

Wie man sieht, reagiert die WP immer dann, wenn die DHW-Temperatur unter 50 °C geht. Wie das mit "Hysterese 3" einhergeht, ist mir ehrlich gesagt nicht ganz klar. Vielleicht interner Zielwert von 52 °C mit Ausschaltautomatik bei +3 °C und Einschaltautomatik bei -3°C?

Naja, schön wäre jedenfalls, wenn ich auf Knopfdruck sagen könnte: DHW aufheizen JETZT! -> Interessant bei z.B. PV-Überschuss. Bisher fällt mir dazu nur der Desinfektionsmodus ein, da macht er das dann auch tatsächlich, ist dann aber erst mal eine Stunde im Desinfektionsmodus...

Anderes Problem: Man sieht hier auch, dass mein Warmwasserspeicher extrem schnell kalt wird: ca. 1 °C pro Stunde ab z.B. 9 Uhr. Woran könnte das liegen? 
Das Problem besteht nicht bzw. nicht in diesem Maße, wenn die Durchflussrate bei ca. 10 ltr/min ist und die Anlage mit relativ konstanter kompressordrehzahl durchläuft , z.B. zwischen 15 und 18 Uhr. Ich weiß, das ist ein anderes Thema, aber wo ich hier jetzt schon mal ein DHW-Screenshot hochlade... ;-)