basti242/homeassistant_lg_therma_v_modbus #17: Kommunikation / Steuerung

TheNablaOperator, 2025-01-01:
Grüß euch.

Danke für das wunderbare Tool.
Ich habe schon ein paar Diskussionen gelesen und werde aber nicht ganz schlau. Da es nicht ganz dazu passt mache ich ein neues auf.

Aktuell habe ich die Config soweit vorgenommen, denke ich (new to HA). Sieht dann bei mir so aus (HACS habe ich auch schon drinnen, aber ganz weiß ich nicht was hier zu tun ist...):

[image]

Jetzt habe ich schon gelesen, "Nicht verfügbar" womöglich ein Firmware-problem (alte vs. neue Version)? Oder liegt es eher an der Modbus Kommunikation? Da auch alle Schalter ausgegraut sind.

Eine weitere Frage hätte ich noch. In der Installateur-Ebene kann man RMC auch von Master auf Slave stellen, dann sind alle Einstellungen in der Remote praktisch weg, ist das auch notwendig oder reicht die Adressänderung vom Modbus und die DIP-Schalter?

Danke im Voraus.
Grüße Nabla

Gyr0sK0mplett, 2025-01-02:
Da findet ganz offensichtlich keine Kommunikation statt.
Welchen Waveshare nutzt du? Die WiFi Variante funktionierte zumindest bei uns überhaupt nicht, keinerlei Kommunikation.
Ansonsten wirklich penibel alle Schritte von Basti durchgehen, Config überprüfen. Dann funktioniert es einwandfrei.

HACS ist ein anderes Thema. Für Grundlagen empfehle ich das Forum von Simon42.

basti242, 2025-01-02:
Würde ich auch empfehlen. Gehe jeden Schritt nochmal genau durch. 
Kannst auch die relevanten Einstellungen mal zeigen.

TheNablaOperator, 2025-01-04:
Grüß euch. Ja liegt an der Kommunikation. Aktuell per Wifi. OK, dann werde ich wohl ein Kabel legen. Danke vorerst.

basti242, 2025-01-04:
Ja leg mal ein Kabel. 
Irgendeiner hat es aber mit WLAN schon hinbekommen. Sollte vielleicht mal dokumentiert werden. 
Ich schaue mal ob ich das finde. 

Gruß
Basti

Gyr0sK0mplett, 2025-01-04:
> Grüß euch. Ja liegt an der Kommunikation. Aktuell per Wifi. OK, dann werde ich wohl ein Kabel legen. Danke vorerst.

Nur zur Sicherheit, bei mir funktionierte es grundsätzlich nicht mit dem WiFi Waveshare, weder Kabel noch WiFi. Hab mir dann den Standard ETH geholt, damit ging es sofort.

TheNablaOperator, 2025-01-06:
OK, Danke. Ich muss noch vorher den Mischerregler fertig einbinden, dann kommt noch mal die WP dran... Ich werde da mal was versuchen... Hoffe es klappt. Könnte aber ein wenig dauern...

TheNablaOperator, 2025-01-19:
Grüß euch. 
Erstmal das Update:
Kommunikation mit Waveshare RS485-ETH (Rail, PoE) läuft. :) Zwar noch nicht per PoE Versorgung, die kommt aber demnächst. Hatte kein Kabel mehr ^^

Was ich versucht habe. Ich hatte noch ein paar RS485 Module und ich versuchte die Kommunikation einmal über Raspi (NodeRed) bzw. ESP32 (Modbus Gateway). Beim ESP, habe ich - nachdem ich das Pegelproblem gelöst hatte - zumindest eine Antwort von LG bekommen, ich bin mir nicht sicher ob es am WiFi lag, dass es nicht klappte. Ich bleibe da aber dran, vielleicht wird es was...

HACS, so dachte ich dass ich alles gemacht habe. Ich gehe zur Sicherheit noch einmal alles durch, sieht trotz Kommunikation noch nicht wie bei euch aus...

Noch etwas anderes. Wenn die Pumpe nicht läuft, gibt euch der Durchflusssensor auch 5l/min? 0 war es nicht... Wegen den tiefen Temperaturen habe ich jetzt mal nicht den Kugelhahn abgedreht...

Grüße, Nabla

TheNablaOperator, 2025-01-29:
Hallo. 
Entities passen nun, shame on me... 

Der Durchflusssensor gibt trotzdem immer noch 5l/min aus, wenn die Pumpe steht. 
Und, muss noch etwas gesetzt werden, damit die Schalter funktionieren? Wenn ja, dann habe ich da etwas nicht gecheckt. Ansonsten sind alle Werte wunderbar da... 
Danke hierfür!

basti242, 2025-02-03:
5l/min ist nicht viel, vielleicht ist das eine Frostschutzfunktion?

TheNablaOperator, 2025-02-03:
Hallo Basti.
Hmmm denke nicht, habe den Kugelhahn zugedreht und es war immer noch bei 5l/min. Lt. Anzeige im Panel: 0l/min. Geht es bei dir/euch auf 0l/min in HA?

Kannst du die WP bei dir ein- bzw. ausschalten per Modbus? Wie gesagt, die Schalter schreiben anscheinend nicht rüber, die nehmen nur den Status der WP. 

Danke im Voraus.

TheNablaOperator, 2025-02-03:
> Ja leg mal ein Kabel. Irgendeiner hat es aber mit WLAN schon hinbekommen. Sollte vielleicht mal dokumentiert werden. Ich schaue mal ob ich das finde.
> 
> Gruß Basti

Hab es bei dir im Repo gefunden. @fposch84 in Discussion #13

basti242, 2025-02-03:
> Hallo Basti.
> Hmmm denke nicht, habe den Kugelhahn zugedreht und es war immer noch bei 5l/min. Lt. Anzeige im Panel: 0l/min. Geht es bei dir/euch auf 0l/min in HA?
>
> Kannst du die WP bei dir ein- bzw. [...]

Bei mir geht's auch nur auf 5l/min runter.
Warum schreiben die Schalter nicht. Die Automatisierungen in der yaml sind noch drin?

TheNablaOperator, 2025-02-04:
> Bei mir geht's auch nur auf 5l/min runter. Warum schreiben die Schalter nicht. Die Automatisierungen in der yaml sind noch drin? Bei mir gehen natürlich alle Schalter.

OK danke, dann ist 5l/min wohl der Minimumwert.

Wenn du die YAML im /integrations Verzeichnis meinst, ja. Habe zur Sicherheit auch das File noch einmal reinkopiert.
Vielleicht eine doofe Frage: Reichen die Automations in dieser Datei, oder braucht es bei den Entities im Dashboard auch noch die Automation-Abschnitte?

TheNablaOperator, 2025-02-25:
Grüß euch... Ganz rund läuft es noch nicht. 
Wie laufen bei euch die Abfragen der Modbus-Register? Die Werte der Sensoren werden sauber übertragen. Ein Abschalten per Hauptschalter funktioniert ab und an. Mir ist aber aufgefallen, dass der Zeitstempel von wann der Schalter Status kommt, recht lange in der Vergangenheit liegt (z.B. 10h). Wenn ich mir dann die Automation [Heatpump]... ansehe, wurde die vor Ewigkeiten ausgeführt aber nicht alle 60s (sollte sie doch tun oder)?
Danke.

basti242, 2025-02-26:
Moin, 

wenn die Schnittstelle läuft und du Werte empfängst oder senden kannst, sollte auch das Auschalten der Anlage immer funktionieren.
Bisschen mehr Informationen oder Screenshots wären hilfreich. Was soll man mit den 10h anfangen? Welche Zeit erwartest du? Wann hast du den Schalter das letzte mal betätigt?
Welche Automation soll alle 60s getriggert werden und warum sollte sie.
Bitte

TheNablaOperator, 2025-02-28:
Servus. 
Also ich bekomme die Werte alle gesendet (Sensoren). Wenn ich nun die Anlage per HA abschalte, dauert es etwas (wie du in der Anleitung geschrieben hast). Manchmal passiert auch nichts, und beim Reload des Dashboards bleibt der Alte Zustand. Wobei ich hatte auch schon dass nach einem Einschaltversuch per HA, nichts passiert (3min), 10min später war die Anlage dann doch ein....
Also ganz verstehe ich noch nicht ob HA ein Problem hat, oder Modbus, oder das Netzwerk noch Probleme machen. Mittlerweile hängt HA und der Modbus Converter sogar am gleichen Switch.

Damit ich das richtig verstehe, werden die Modbus-Befehle per Zeitstempel aktualisiert oder vom Modbus selbst getriggert? Hier meinte ich  eine gesehen zu haben die alle 60s getriggert wird...?

Wenn ich dein Kommentar richtig verstehe, ist der Zeitstempel wann sich der Schalter geändert hat...?

[image]

[image]

Grüße!