basti242/homeassistant_lg_therma_v_modbus #18: Interface via Serial (USB_Adapter)

isarrider, 2025-01-05:
Hi,

when my Therma V R290 is here,
I will also test if it works via Serial:

modbus:
  - name: "lg_heatpump"
    type: serial
    baudrate: 9600
    bytesize: 8
    method: rtu
    parity: N
    port: /dev/ttyUSB0
    stopbits: 1

Or did anyone do that yet?

BR,
Alex

closed 2025-02-03 by basti242

isarrider, 2025-11-09:
I am trying now to interface the heatpump to my RS485 USB Dongle...
But somehow it wont work, I attached the config - idea?

isarrider, 2025-11-09:
[modbus_lg_heatpump.yaml](https://github.com/user-attachments/files/23437519/modbus_lg_heatpump.yaml)

maximilianadolf, 2025-11-09:
Wenn du schon über EVCC auch mit der LG verbunden bist, dann musst du natürlich einen Modbus Proxy einrichten. Zwei Master im Modbus wird sich sehr oft beißen. In EVCC geht das ziemlich einfach einzurichten und benötigt kein weiteres Addon wie in HA.

isarrider, 2025-11-09:
Hallo Auchmünchner,

danke Dir für die Antwort...
Ich habe das device in Docker gemountet, es gibt es...
Ich teste mal heute abend, ob ich es direkt von /dev/ttyUSB1 (ist es bei mir) sehe...

(Aber es hält mich fit, ca heute schon 25x in den Keller gerannt ;))

VG;
Alex

basti242, 2025-11-10:
> @basti242 https://github.com/basti242
> ggf in die Docu aufnehmen, daß ein USB RS485 Stick auch geht?

Kannst den USB RS485 Stick mal mit in die Doku nehmen. Im Code würde ich ihn aber nur als Variante auskommentiert reinbringen. Ist ja denke ich eher ne Sonderlocke.

isarrider, 2025-11-10:
Ok, ich pack das rein...
Weiß gar nicht ob das so ne Sonderlocke ist,
ich kenn doch ein paar, die so Modbus machen...
PR kommt...