basti242/homeassistant_lg_therma_v_modbus #22: Einige Entitäten nicht verfügbar

TomHeckl, 2025-01-26:
Hi Basti,
erstmal vielen Dank für die erstklassige Anleitung.  Ich hatte zwar anfangs einige Probleme, die sich aber in Luft aufgelöst haben, nachdem ich die Wärmepumpe an und wieder aus gemacht habe.

Ich habe nur noch zwei Probleme:
1.  Die Schalter für SilentMode und An/Aus Wärmepumpe haben keine Wirkung. Die Wärmepumpe reagiert nicht darauf.
2. . Einige Entitäten sind nicht verfügbar.
- sensor.vorlauf_hk1_esp
- sensor.vorlauf_hk2_esp
- sensor.temp_before_vaporiser
- sensor.temp_liquid_gas
- switch.zirkulationspumpe
- binary_sensor.hp_flow_too_low
- climate.hp_heating_climate
- binary_sensor.hp_heating_mode
- sensor.hp_mode
- sensor.hp_outdoor_air_temp
- binary_sensor.hp_pump_status
- binary_sensor.hp_silent_status
- climate.hp_water_climate

Im Protokoll werden auch dazu ein paar Fehler ausgegeben. Ich komme hier aber leider nicht weiter.

Ich habe eine LG Therma V (HM123MR.U34) WP mit einem RS485toETH (B) von Waveshare. Home Assistant läuft in der Version 2025.1.4

Hier ein Auszug aus dem Protokoll:

[image]

Es wäre super, wenn du mir einen Hinweis geben könntest, wie und wo ich mich auf die Fehlersuche machen soll / kann.

Danke und viele Grüße
Tom

basti242, 2025-01-26:
Hi Tpm, 

die ESP Sensoren sind nicht Teil der Heizung. 
switch.zirkulationspumpe auch nicht. 

Grundsätzlich solltest du die Entitäten in der "modbus_lg_heatpump.yaml" finden. Wenn sie da nicht drin sind, sollte es sie auch nicht geben. Beispiel: binary_sensor.hp_flow_too_low. Die Entität sagt mir gar nix.

Was hast du für ein Model? Baujahr und Typ usw.

Ich werde die Beispiele bei Gelegenheit mal überarbeiten.

Gruß
Basti

TomHeckl, 2025-01-26:
Hallo Basti,

vielen Dank für die Antwort am Sonntag. 😊

Ich habe ein LG Therma V, die im Mai 2023 eingebaut wurde.

 

Ich hab heute die Integration fertiggestellt und kurz nachdem ich dir geschrieben habe, ging die Anlage auf Störung. Evtl. hängt das mit dem Ein/Aus-Schalter zusammen der augenscheinlich keine Wirkung hat, aber evtl. dennoch irgendwelche Register manipuliert…???

 

Ich musste dann einen Reset machen (Anlage vom Netz nehmen) und wieder hochfahren. Die Heizung läuft wieder, aber nun ist die Integration auf „rot“.

 

 

Irgendwas scheint da nicht zu passen, aber ich hab keinen Plan, wo ich das Suchen anfangen soll. 

 

VG
Tom

TomHeckl, 2025-01-26:
p.MsoNormal, li.MsoNormal, div.MsoNormal { margin: 0; font-size: 12pt; font-family: "Aptos", sans-serif }
a:link, span.MsoHyperlink { color: rgba(0, 0, 255, 1); text-decoration: underline }
span.E-MailFormatvorlage20 { font-family: "Aptos", sans-serif; color: rgba(0, 0, 0, 1) }
.MsoChpDefault { font-size: 11pt }
div.WordSection1 { }
@media only screen {html,body {margin: 0;padding: 5px;width: 100% !important;max-width: 100% !important;box-sizing: border-box !important;-webkit-box-sizing: border-box !important;-moz-box-sizing: border-box !important;-webkit-text-size-adjust: 100%;-ms-text-size-adjust: 100%;}body[style] {min-width: auto !important;width: 100% !important;max-width: 100% !important;}*,*:before,*:after {box-sizing: inherit !important;-webkit-box-sizing: inherit !important;-moz-box-sizing: inherit !important;-webkit-text-size-adjust: inherit !important;-ms-text-size-adjust: inherit !important;}table {margin: 0;padding: 0;width: 100% !important;}table[style] {min-width: auto !important;width: 100% !important;max-width: 100% !important;}img {margin: 0;padding: 0;max-width: 100% !important;height: auto;}img[style] {min-width: auto !important;max-width: 100% !important;height: auto !important;}}Waveshare neu gestartet und jetzt kommen auch wieder Daten. Scheint aber irgendwie ein fragiles System zu sein...damit meine ich mein System hier. 😀VGTomAm 26. Jan. 2025 17:37 schriebVon: ***@***.***: ***@***.***: AW: [basti242/homeassistant_lg_therma_v_modbus] Einige Entitäten nicht verfügbar (Issue #22)Hallo Basti,vielen Dank für die Antwort am Sonntag. 😊Ich habe ein LG Therma V, die im Mai 2023 eingebaut wurde.Ich hab heute die Integration fertiggestellt und kurz nachdem ich dir geschrieben habe, ging die Anlage auf Störung. Evtl. hängt das mit dem Ein/Aus-Schalter zusammen der augenscheinlich keine Wirkung hat, aber evtl. dennoch irgendwelche Register manipuliert…???Ich musste dann einen Reset machen (Anlage vom Netz nehmen) und wieder hochfahren. Die Heizung läuft wieder, aber nun ist die Integration auf „rot“.Irgendwas scheint da nicht zu passen, aber ich hab keinen Plan, wo ich das Suchen anfangen soll. VGTomVon: basti242 <notification
 ***@***.***> Gesendet: Sonntag, 26. Januar 2025 17:14An: basti242/homeassistant_lg_therma_v_modbus ***@***.***>Cc: TomHeckl ***@***.***>; Author ***@***.***>Betreff: Re: [basti242/homeassistant_lg_therma_v_modbus] Einige Entitäten nicht verfügbar (Issue #22)Hi Tpm,die ESP Sensoren sind nicht Teil der Heizung.switch.zirkulationspumpe auch nicht.Grundsätzlich solltest du die Entitäten in der "modbus_lg_heatpump.yaml" finden. Wenn sie da nicht drin sind, sollte es sie auch nicht geben. Beispiel: binary_sensor.hp_flow_too_low. Die Entität sagt mir gar nix.Was hast du für ein Model? Baujahr und Typ usw.Ich werde die Beispiele bei Gelegenheit mal überarbeiten.GrußBasti

basti242, 2025-02-03:
kontrolliere bitte die "[modbus_lg_heatpump.yaml](https://github.com/basti242/homeassistant_lg_therma_v_modbus/blob/main/modbus_lg_heatpump.yaml). Wahrscheinlich stimmt da was nicht, und wenn es nur eine falsche Einfückung ist. Yaml ist da durchaus zickig.

closed 2026-10-19 by basti242