basti242/homeassistant_lg_therma_v_modbus #23: Fehler Ch242 (Zentralsteuerung) nach Umstellung auf Modbus

thitcher, 2025-01-26:
Hallo zusammen.
Ich habe heute versucht meine LG therma V auf Modbus umzustellen.
Ich hatte bei der Installation vorletztes Jahr bereits ein LAN Kabel mit reingezogen und dieses am 3trd Party Controller angeschlossen. Allerdings habe ich die Switches nicht gesetzt gehabt und das heute nachgeholt.
Also die Schalter 1 und 2 von Schalterbank 1 auf on gesetzt.
Dann den waveshare angeschlossen und in der Steuerung die Zentraladresse auf 01 gestellt. Modbus war schon auf 01.
Nun habe ich das Problem, daß meine thinq app nicht mehr funktioniert.
Die Steuereinheit (also das Display) schon.
Fehler CH 242.
Ist das normal das die App nicht mehr funktioniert?
Ich dachte ich hätte gelesen das trotzdem weiterhin alles wie gewohnt funktioniert.

thitcher, 2025-09-26:
Das wars!
Jetzt geht es wieder, ich vielen Dank 👌🏻

RedGreenChilli, 2025-11-13:
Hi @maximilianadolf mit diesen Einstellungen geht meine WP auf Störung sobald das Modbus GW nicht mehr läuft. Hast du das gleiche Problem?