basti242/homeassistant_lg_therma_v_modbus #28: COP calculation from entalphy

andreads28, 2025-03-13:
Hello dears,

I’m trying to calculate the COP by having R32 properties in a dictionary yaml. Unfortunately the temperature and pressure interpolation do not find the right properties value to obtain the entalphy. Can you help? I’ve used the sensors in this code and deepseek helped me.
Code attached.
It is in txt but is .py extension in reality.
Can you help me?
I’ve added it as custom integration.

[sensor.txt](https://github.com/user-attachments/files/19228829/sensor.txt)

32u-nd, 2025-08-05:
Hi! Your post is a few months old, however this may help. There is no need to calculate COP based on the enthalpy. You need:

a) temperature difference of heat pump inlet and outlet (K) = (modbus 30004 - modbus 30003)/10
b) flow rater (l/min) = (modbus 30009)/10

Water at 2 bar pressure and 40 °C (my average system setting) has a heat capacity factor of 69.1 W/K per 1 l/min, so

heat power = 69.1 * a * b  (W)

COP = heat power / electricity consumption

BR, Andreas

andreads28, 2025-08-05:
> There are some energy values available from the MODBUS. You can also find them in the documentation from Basti in the WIKI.
>
> I have implemented those for me and will observe those for now and compare to the values reported in the control unit.
> Since I only started reading out values since the weekend, I focus right now on the water values "hp_ENERGY_ACCUMULATIVE_DHW_low" on input register 45:
>
> ` - name: "hp_ENERGY_INSTANTANEOUS_TOTAL_high"
> unique_id: "modbus.hp_ENERGY_INSTANTANEOUS_TOTAL_high"
> precision: 2
> scan_interval: 10
> address: 36
> slave: !secret lg_heatpump_modbus_slave
> unit_of_measurement: Wh
> input_type: input
> device_class: energy
>
>   - name: "hp_ENERGY_INSTANTANEOUS_TOTAL_low"
>     unique_id: "modbus.hp_ENERGY_INSTANTANEOUS_TOTAL_low"
>     precision: 2
>     scan_interval: 10
>     address: 37
>     slave: !secret lg_heatpump_modbus_slave
>     unit_of_measurement: Wh
>     input_type: input
>     device_class: energy
>
> [... 82 lines cut ...]
>     input_type: input
>     device_class: gas #just to test if I can integrate this into my energy dashboard :-)
>     state_class: total`
>
> I have a Thermondo installation and hence also a SDM72D for electic energy of the complete installation. I'll currently record everything and will compare values when sufficient data is available.

Many thanks to have discovered these ‘secret’ addresses!

Let us to know also what ‘total high’ and ‘total low’ mean in order to better understand if these values are the ones of the control panel!
BR
Andrea

andreads28, 2025-10-19:
Hello! I’ve started to try the energy values to understand how to catch the thermal energy produced. 
I’ve discovered that hp_ENERGY_ ISTANTANEOUS_TOTAL measures the instantaneous electrical power spent by the HP.
From the values identified, the hp_ENERGY_ACCUMULATIVE_TOTAL_low should measure the thermal power generated. However this sensor after more or less 10 days goes down to a negative value without any change in HP settting (always heat mode).
Does anyone know how to gather the heat power released by the HP? Or at least how does the internal statics work? Attached the sensor statistic.

> 

[image]

There are some energy values available from the MODBUS. You can also find them in the documentation from Basti in the WIKI.
> I have implemented those for me and will observe those for now and compare to the values reported in the control unit. Since I only started reading out values since the weekend, I focus right now on [...]